BlackZork/mqmgateway #4: Unittests is fails (2 tests)

dwrobel, 2022-03-01:
Unittests seems to be failing on Fedora 35 aarch64 platform as following:
```
tests is a Catch v2.13.5 host application.
Run with -? for options

----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
-------------------------------------------------------------
When modbus network is connected object state should be published immediately
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
-------------------------------------------------------------
/home/dw/projects/mqmgateway/unittests/single_register_tests.cpp:58
..............................................................................................................................................................................................................................................
.............................................................

/home/dw/projects/mqmgateway/unittests/mockedserver.hpp:79: FAILED:
  REQUIRE( is_published == true )
with expansion:
  false == true
with message:
  Checking for publish on test_switch/state

----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

-------------------------------------------------------------
If broker is restarted all mqtt objects should be republished
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
-------------------------------------------------------------
/home/dw/projects/mqmgateway/unittests/single_register_tests.cpp:105
..............................................................................................................................................................................................................................................
.............................................................

/home/dw/projects/mqmgateway/unittests/mockedserver.hpp:79: FAILED:
  REQUIRE( is_published == true )
with expansion:
  false == true
with message:
  Checking for publish on test_switch/state

==============================================================================================================================================================================================================================================
=============================================================
test cases:  42 |  40 passed | 2 failed
assertions: 189 | 187 passed | 2 failed
```

For the record list of versions of dependency libraries:
```
$ ldd tests | awk '{print $3}' | grep /lib64 | xargs rpm -qf | sort | uniq
boost-atomic-1.76.0-4.fc35.aarch64
boost-chrono-1.76.0-4.fc35.aarch64
boost-date-time-1.76.0-4.fc35.aarch64
boost-filesystem-1.76.0-4.fc35.aarch64
boost-log-1.76.0-4.fc35.aarch64
boost-program-options-1.76.0-4.fc35.aarch64
boost-regex-1.76.0-4.fc35.aarch64
boost-system-1.76.0-4.fc35.aarch64
boost-thread-1.76.0-4.fc35.aarch64
c-ares-1.17.2-1.fc35.aarch64
glibc-2.34-25.fc35.aarch64
libgcc-11.2.1-9.fc35.aarch64
libicu-69.1-2.fc35.aarch64
libmodbus-3.0.8-5.fc35.aarch64
libstdc++-11.2.1-9.fc35.aarch64
mosquitto-2.0.14-1.fc35.aarch64
openssl-libs-1.1.1l-2.fc35.aarch64
yaml-cpp-0.6.3-5.fc35.aarch64
zlib-1.2.11-30.fc35.aarch64
```

BlackZork, 2022-03-04:
Works here on arch. 
Are they still failing if you increase timeout in waitForPublish for example to 1 second?

BlackZork, 2023-05-26:
Timeouts fixed in [5de54c8](https://github.com/BlackZork/mqmgateway/commit/5de54c83a6be9ea9184a98e557834595dedc54b5)

closed 2023-05-26 by BlackZork