BlackZork/mqmgateway #5: build

VillageEngineer, 2022-03-22:
Hello, I can't build your project. What am I doing wrong?

pi@raspberrypi:~/mqmgateway $ cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -S /home/pi/mqmgateway/ -B /home/pi/mqmgateway/build/
-- RapidJSON found. Headers: /usr/include
-- EXPRTK_INCLUDE_DIR => EXPRTK_INCLUDE_DIR-NOTFOUND
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
EXPRTK_INCLUDE_DIR
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv

-- Configuring incomplete, errors occurred!
See also "/home/pi/mqmgateway/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/mqmgateway/build/CMakeFiles/CMakeError.log".

BlackZork, 2022-03-22:
Looks like a bug in cmake.
mqmgateway uses exprtk for advanced conversion. It is an optional converter and project should compile without it. I will look at it.

Jihadist, 2022-03-22:
@VillageEngineer please post an output of ```ls /home/pi/mqmgateway/exprconv```

VillageEngineer, 2022-03-22:
@Jihadist 
pi@raspberrypi:~/mqmgateway/exprconv $ ls -l
итого 48
-rw-r--r-- 1 pi pi 14031 мар 22 10:25 CMakeCache.txt
drwxr-xr-x 6 pi pi  4096 мар 22 10:36 CMakeFiles
-rw-r--r-- 1 pi pi  1502 мар 22 10:25 cmake_install.cmake
-rw-r--r-- 1 pi pi   117 мар 20 16:37 CMakeLists.txt
-rw-r--r-- 1 pi pi  1898 мар 20 16:37 expr.hpp
-rw-r--r-- 1 pi pi  4635 мар 22 10:25 Makefile
-rw-r--r-- 1 pi pi   215 мар 20 16:37 plugin.cpp
-rw-r--r-- 1 pi pi   444 мар 20 16:37 plugin.hpp

pi@raspberrypi:~/mqmgateway/exprconv $ make
[ 50%] Building CXX object CMakeFiles/exprconv.dir/plugin.o
In file included from /home/pi/mqmgateway/exprconv/plugin.cpp:1:
/home/pi/mqmgateway/exprconv/plugin.hpp:5:10: fatal error: libmodmqttconv/converterplugin.hpp: Нет такого файла или каталога
 #include "libmodmqttconv/converterplugin.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/exprconv.dir/build.make:63: CMakeFiles/exprconv.dir/plugin.o] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/exprconv.dir/all] Ошибка 2
make: *** [Makefile:84: all] Ошибка 2

Jihadist, 2022-03-22:
`find_path(EXPRTK_INCLUDE_DIR "exprtk.hpp")`
Looks like /home/pi/mqmgateway/exprconv doesnt containe necessary header

VillageEngineer, 2022-03-22:
I builded exprtk.hpp and copy in /home/pi/mqmgateway/exprconv   

```
pi@raspberrypi:~/mqmgateway $ ls -l exprconv/
итого 1488
-rw-r--r-- 1 pi pi   14031 мар 22 10:25 CMakeCache.txt
drwxr-xr-x 6 pi pi    4096 мар 22 12:37 CMakeFiles
-rw-r--r-- 1 pi pi    1502 мар 22 10:25 cmake_install.cmake
-rw-r--r-- 1 pi pi     117 мар 20 16:37 CMakeLists.txt
-rw-r--r-- 1 pi pi    1898 мар 20 16:37 expr.hpp
-rw-r--r-- 1 pi pi 1472878 мар 22 12:41 exprtk.hpp
-rw-r--r-- 1 pi pi    4635 мар 22 10:25 Makefile
-rw-r--r-- 1 pi pi     215 мар 20 16:37 plugin.cpp
-rw-r--r-- 1 pi pi     444 мар 20 16:37 plugin.hpp
```

```
~/mqmgateway $ cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -S /home/pi/mqmgateway/ -B /home/pi/mqmgateway/build/
-- RapidJSON found. Headers: /usr/include
-- EXPRTK_INCLUDE_DIR => EXPRTK_INCLUDE_DIR-NOTFOUND
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
EXPRTK_INCLUDE_DIR
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv
   used as include directory in directory /home/pi/mqmgateway/exprconv

-- Configuring incomplete, errors occurred!
See also "/home/pi/mqmgateway/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/mqmgateway/build/CMakeFiles/CMakeError.log".
```

Jihadist, 2022-03-22:
find_path ignores exprconv directory by default because it's non default directory :), look at https://cmake.org/cmake/help/latest/command/find_path.html hints and paths

BlackZork, 2022-03-22:
I fixed build without exprtk library. It should now build fine without it.

If you want to use exprtk expressions then you have to install exprtk - if there is package for it for rpi. If not, just copy exprtk.hpp to /usr/include and mqmgateway should find it and build its optional converter.

VillageEngineer, 2022-03-22:
@BlackZork   Thanks, it worked. But after I make a mock I get this:
```

`[ 23%] Building CXX object libmodmqttsrv/CMakeFiles/modmqttsrv.dir/modmqtt.cpp.o
In file included from /usr/local/include/boost/smart_ptr/detail/shared_count.hpp:44,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:32,
                 from /usr/local/include/boost/smart_ptr/make_shared_object.hpp:16,
                 from /usr/local/include/boost/smart_ptr/make_shared.hpp:15,
                 from /usr/local/include/boost/make_shared.hpp:15,
                 from /usr/include/boost/dll/import.hpp:15,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/utility/addressof.hpp:51:23: error: redefinition of ‘template<class T> T* boost::addressof(T&)’
 template<class T> T * addressof( T & v )
                       ^~~~~~~~~
In file included from /usr/include/boost/dll/import.hpp:12,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/include/boost/core/addressof.hpp:38:1: note: ‘template<class T> constexpr T* boost::addressof(T&)’ previously declared here
 addressof(T& o) BOOST_NOEXCEPT
 ^~~~~~~~~
In file included from /usr/local/include/boost/system/error_code.hpp:19,
                 from /usr/include/boost/dll/detail/system_error.hpp:12,
                 from /usr/include/boost/dll/shared_library.hpp:20,
                 from /usr/include/boost/dll/import.hpp:16,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/utility/enable_if.hpp:28:10: error: redefinition of default argument for ‘class T’
   struct enable_if_c {
          ^~~~~~~~~~~
In file included from /usr/include/boost/dll/import.hpp:13,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/include/boost/core/enable_if.hpp:32:21: note: original definition appeared here
   template <bool B, class T = void>
                     ^~~~~
In file included from /usr/local/include/boost/system/error_code.hpp:19,
                 from /usr/include/boost/dll/detail/system_error.hpp:12,
                 from /usr/include/boost/dll/shared_library.hpp:20,
                 from /usr/include/boost/dll/import.hpp:16,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/utility/enable_if.hpp:33:10: error: redefinition of ‘struct boost::enable_if_c<false, T>’
   struct enable_if_c<false, T> {};
          ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/boost/dll/import.hpp:13,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/include/boost/core/enable_if.hpp:38:10: note: previous definition of ‘struct boost::enable_if_c<false, T>’
   struct enable_if_c<false, T> {};
          ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/boost/system/error_code.hpp:19,
                 from /usr/include/boost/dll/detail/system_error.hpp:12,
                 from /usr/include/boost/dll/shared_library.hpp:20,
                 from /usr/include/boost/dll/import.hpp:16,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/utility/enable_if.hpp:36:10: error: redefinition of default argument for ‘class T’
   struct enable_if : public enable_if_c<Cond::value, T> {};
          ^~~~~~~~~
In file included from /usr/include/boost/dll/import.hpp:13,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/include/boost/core/enable_if.hpp:40:25: note: original definition appeared here
   template <class Cond, class T = void>
                         ^~~~~
In file included from /usr/local/include/boost/system/error_code.hpp:19,
                 from /usr/include/boost/dll/detail/system_error.hpp:12,
                 from /usr/include/boost/dll/shared_library.hpp:20,
                 from /usr/include/boost/dll/import.hpp:16,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/utility/enable_if.hpp:39:10: error: redefinition of ‘struct boost::lazy_enable_if_c<B, T>’
   struct lazy_enable_if_c {
          ^~~~~~~~~~~~~~~~
In file included from /usr/include/boost/dll/import.hpp:13,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/include/boost/core/enable_if.hpp:44:10: note: previous definition of ‘struct boost::lazy_enable_if_c<B, T>’
   struct lazy_enable_if_c {
[... 167 lines cut ...]
     path& append(InputIterator begin, InputIterator end)
           ^~~~~~
/usr/local/include/boost/filesystem/path.hpp:315:11: note:   template argument deduction/substitution failed:
/home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:232:33: note:   candidate expects 2 arguments, 1 provided
         current_path.append(name);
                                 ^
In file included from /usr/include/boost/dll/detail/posix/program_location_impl.hpp:13,
                 from /usr/include/boost/dll/detail/posix/path_from_handle.hpp:12,
                 from /usr/include/boost/dll/detail/posix/shared_library_impl.hpp:13,
                 from /usr/include/boost/dll/shared_library.hpp:26,
                 from /usr/include/boost/dll/import.hpp:16,
                 from /home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:4:
/usr/local/include/boost/filesystem/path.hpp:321:11: note: candidate: ‘template<class InputIterator> boost::filesystem::path& boost::filesystem::path::append(InputIterator, InputIterator, const codecvt_type&)’
[... cut ...]
/home/pi/mqmgateway/libmodmqttsrv/modmqtt.cpp:232:33: note:   candidate expects 3 arguments, 1 provided
         current_path.append(name);
                                 ^
make[3]: *** [libmodmqttsrv/CMakeFiles/modmqttsrv.dir/build.make:154: libmodmqttsrv/CMakeFiles/modmqttsrv.dir/modmqtt.cpp.o] Ошибка 1
make[2]: *** [CMakeFiles/Makefile2:199: libmodmqttsrv/CMakeFiles/modmqttsrv.dir/all] Ошибка 2
make[1]: *** [CMakeFiles/Makefile2:291: unittests/CMakeFiles/tests.dir/rule] Ошибка 2
make: *** [Makefile:203: tests] Ошибка 2`

```

BlackZork, 2022-03-23:
What is your compiler and boost version?

VillageEngineer, 2022-03-24:
Source: boost-defaults (1.67.0.1)
Version: 1.67.0.1+b1
Depends: libboost1.67-dev

cmake version 3.16.3
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

BlackZork, 2022-03-24:
GCC 10.1 and boost 1.70.0 where the first versions that I used at start of this project. I suspect that boost is too old.

VillageEngineer, 2022-03-24:
ok. Thank you

closed 2022-09-19 by BlackZork