BlackZork/mqmgateway #8: Converter Not found

rungg12, 2022-07-27:
systemctl status modmqttd.service says:
Converter std.int32() not found

Do I have to install a extra package?

BlackZork, 2022-07-27:
You need to setup LD_LIBRARY_PATH in your service file, so main exe can find (install_prefix)/lib/modmqttd/stdconv.so

rungg12, 2022-07-27:
I set the path to the build folder where the "stdconv.so" is located, but it doesnt solve the problem

BlackZork, 2022-07-27:
Sorry, wrong advice. Set modmqttd.converter_search_path in config.yaml

rungg12, 2022-07-27:
i put it in the first line of "modmqttd.yaml" with the path to the folder, but it still doesnt work

BlackZork, 2022-07-27:
You can run it under strace:
`strace ./modmqttd/modmqttd --config modmqttd/config.yaml 2>&1 |grep stdconv`

and see where it tries to find this file:

`statx(AT_FDCWD, "../build/stdconv/stdconv.so", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT, STATX_TYPE|STATX_MODE, 0x7ffdc74305a0) = -1 ENOENT (No such file or directory)`

rungg12, 2022-07-27:
thank you very much

closed 2022-09-19 by BlackZork