BlackZork/mqmgateway #13: Handling of unsigned integers

molnarg, 2022-12-17:
It looks like currently, all registers are handled as unsigned numbers. In some use cases, I have signed integers, and option to handle them as such would be welcome. Or maybe a plugin to convert unsigned to signed. Thanks!

BlackZork, 2022-12-17:
There is std.int32 converter, but its name is misleading - it supports merging two registers into a single u_int32 value, not converting uint to int. I will rename it to std.uint32.

To get a signed value you can just substract 65536 from register value. I just added unittest example 
 in b5f17ac03efc038c32716887c5567c462b856e84.

molnarg, 2022-12-19:
Thanks! It's a bit more complicated though. Only need to substract 65535 if the highest bit is 1, otherwise it shouldn't be done. Can I do it with an expr.evaluate call?

molnarg, 2022-12-19:
I could do it this way:

```
converter: 'expr.evaluate("R0 < 32768 ? R0 / 10 : (R0 - 65536) / 10")'
```

Feels a lot like a hack, so a native feature would still be welcome, but for my little side project more than enough :)

BlackZork, 2022-12-19:
Ok, I will leave it open until adding simple cast to int16_t to stdconv.

BlackZork, 2023-05-28:
Added int16 converter in https://github.com/BlackZork/mqmgateway/commit/4444b8a4240f8e184970fc31dc19bcfde08c366b

closed 2023-05-28 by BlackZork