BlackZork/mqmgateway #15: Feature request: converter support for commands

molnarg, 2022-12-19:
When exposing a register over MQTT, and the reading operation has a conversion, it's natural that the same conversion needs to be made in the reverse direction when writing that register.

In my specific case, the register exposes a temperature in units of 0.1C, but over the MQTT interace, I'd like to expose units of 1C. Currently, I can do it for read operations, but not for write operations.

BlackZork, 2022-12-19:
The simplest way to do it is to use similar interface as for state topic value conversions. TODO:

- add ICommandConverter interface and implementation. Stdconv, exprconv need to provide implementation for converting value in opposite direction, after converting mqtt payload to uint16_t. It is simpler, because we have uint16_t as input and output value.
- add tests for valid and invalid (e.g. out of range, conversion exceptions) conversions
- documentation

It will take some time.

BlackZork, 2023-05-21:
Implemented in 48cb1cb6

closed 2023-05-21 by BlackZork