BlackZork/mqmgateway #16: Write to register from MQTT topic

pki791, 2023-03-21:
Hi.
The readme states `Allows to read and write to MODBUS registers from MQTT side with custom data conversion`.

Please provide a config example how to properly allow a write from MQTT topic to modbus register.
Thx

BlackZork, 2023-03-21:
See [unit tests for commands](https://github.com/BlackZork/mqmgateway/blob/master/unittests/mqtt_command_tests.cpp).

xspander, 2023-04-05:
Is it possible to write multiple registers? 0x10 command

BlackZork, 2023-04-06:
Read & write to multiple registers is not implemented yet.

BlackZork, 2023-05-26:
Read implmented in [5de54c8](https://github.com/BlackZork/mqmgateway/commit/5de54c83a6be9ea9184a98e557834595dedc54b5).

A summary of changes:
 - new way to define register lists: as register_number + count. This declaration issues single read call to modbus slave. Old way causes multiple read calls to slave.
 - fixed register adressing in config file, now uses human readable format (1 is register 0x0)
 - you can declare poll groups in modbus section, i.e. to pull many registers at once regardless how mqtt topics are declared.
 
Details in README.md, I added some unit tests for new scenarios.

BlackZork, 2023-05-27:
> Is it possible to write multiple registers? 0x10 command

@xspander
Implemented in https://github.com/BlackZork/mqmgateway/commit/e13c1cf2c492a01fe1f9e84857f6a8ef3bfbd1bc. 
You can now specify "count" in command section.

Gateway accepts json list by default, or you can write converter that will create list of registers from published mqtt value.

closed 2023-05-27 by BlackZork