BlackZork/mqmgateway #17: Failed to load configuration from /etc/modmqttd/config.yaml

xspander, 2023-03-22:
Hello, trying to start modmqttd service, it fails with result 'exit-code'. In journalctl show this ` 2023-Mar-22 14:24:54.375589: [INFO]        modmqttd is starting
мар 22 14:24:54 jethubj100 modmqttd[6880]: 2023-Mar-22 14:24:54.378455: [CRITICAL]        Failed to load configuration from /etc/modmqttd/config.yaml
` 
thanks in advance for your help

BlackZork, 2023-03-22:
You need a config file, where MQTT topics are mapped to modbus registers. See example in source code.

pki791, 2023-03-22:
Maybe wrong permissions on the config file or directory? I used it few days ago without a problem.

BlackZork, 2023-03-22:
You can use strace to see if process is trying to open it and a result

xspander, 2023-03-22:
`adminx@jethubj100:/etc/modmqttd$ strace -o trace_output.txt modmqttd
2023-Mar-22 15:06:00.801679: [INFO]     modmqttd is starting
2023-Mar-22 15:06:00.804620: [WARNING]  No config path, trying to read config.yaml from working directory
2023-Mar-22 15:06:00.806826: [CRITICAL] Failed to load configuration from /etc/modmqttd
adminx@jethubj100:/etc/modmqttd$ dir
config.yml  config.yml.backup  trace_output.txt
adminx@jethubj100:/etc/modmqttd$
`
It does not see this file anyway

xspander, 2023-03-22:
Oh, i finded a problem, config file must be "config.yaml"

xspander, 2023-03-22:
Thank for help) Can you change in [README.md] "Copy config.template.yml to /etc/modmqttd/config.yml" to "Copy config.template.yml to /etc/modmqttd/config.yaml"

closed 2023-03-22 by xspander