BlackZork/mqmgateway #30: MQTT Authentication not implemented? username and password not being picked up

dw-1983, 2023-09-22:
Firstly, what I can see from the documentation - great work! I have tried different gateways and none provides the flexibility I see here to bulk read and write multiple registers from/to a modbus slave in combination of mapping this functionality to MQTT topics.

My use case:
- Read sensor data (holding registers) contained in multiple subsequent register-blocks from a Sugar Valley Pool Controller and (same modbus line) a HE200 VFD Inverter for filte rpump motor controlling
- Write holding registers to both components for starting/stopping filtering, setting pump motor speed, etc. I have the special need to write a single register value using Function 16 (0x10, write_multiple_holding_register), as the component does not accept/react to a single register write operation (at least for a special register) when writing a value of 0. Maybe I will raise another issue for this case, currently I can help myself in writing the following register, too, to force a write_multiple.

My current setup:
- I am running a mosquitto broker/server on OpenWRT using the OOTB user/password authentication using passwd file. Connection to the broker runs fine since several years using user/passwd from different clients like tasmota, MQTTX and HA.
- I am running the latest mqmgateway in a docker container using the host network on QNAP NAS stotrage (Container Station)
- The gateway comes up fine in the logs and connects to the broker. The connection is listed in the mosquitto logs of the server broker.
- After connecting, the server closes the connection due to missing authentication.

My mqtt configuration snippet for mqmgateway:

`mqtt:
  client_id: 'modbus'
  refresh: 1s
  broker:
    host: 192.168.2.1
    port: 1883
    keepalive: 60
    username: '<masked username>'
    password: '<masked password>'`

The logs of the gateway look like this:
`2023-Sep-22 11:45:46.199642: [INFO]     modmqttd is starting
2023-Sep-22 11:45:46.248124: [INFO]     Added converter plugin std
2023-Sep-22 11:45:46.327991: [INFO]     Connecting to 192.168.2.208:8899
2023-Sep-22 11:45:46.328252: [INFO]     modbus: connecting
2023-Sep-22 11:45:46.329952: [INFO]     modbus: connected
2023-Sep-22 11:45:46.330161: [INFO]     Waiting for mqtt network to become online
2023-Sep-22 11:45:46.351543: [INFO]     Connecting to 192.168.2.1:1883
2023-Sep-22 11:45:46.351816: [INFO]     Waiting for mqtt network to become online
2023-Sep-22 11:45:46.361430: [INFO]     Connection estabilished
2023-Sep-22 11:45:46.361664: [INFO]     Mqtt conected, sending subscriptions...
2023-Sep-22 11:45:46.361845: [INFO]     Mqtt ready to process messages
2023-Sep-22 11:45:46.361976: [INFO]     Disconnected from mqtt broker, code:The connection was refused.
2023-Sep-22 11:45:46.362084: [INFO]     reconnecting to mqtt broker
2023-Sep-22 11:45:46.449680: [INFO]     Disconnected from mqtt broker, code:The connection was lost.
2023-Sep-22 11:45:46.449902: [INFO]     reconnecting to mqtt broker`

The logs of the broker server look like this:
`Fri Sep 22 13:21:48 2023 daemon.notice mosquitto[30795]: 1695381708: New connection from 192.168.2.202:49910 on port 1883.
Fri Sep 22 13:21:48 2023 daemon.debug mosquitto[30795]: 1695381708: Sending CONNACK to modbus (0, 5)
Fri Sep 22 13:21:48 2023 daemon.notice mosquitto[30795]: 1695381708: Client modbus disconnected, not authorised.`

It looks like the mqtt client does not use credentials at all, as successful user/pass connection attempts look like this:
`Fri Sep 22 13:21:43 2023 daemon.notice mosquitto[30795]: 1695381703: New connection from 192.168.3.178:16281 on port 1883.
Fri Sep 22 13:21:43 2023 daemon.notice mosquitto[30795]: 1695381703: New client connected from 192.168.3.178:16281 as DVES_XXXX (p2, c1, k15, u'USERNAME_MASKED')`

As far as I can see from the code, [mosquitto_username_pw_set](https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_username_pw_set) is not being called prior to [mosquitto_connect](https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_connect).
As I am no C++ developer, I don't trust myself to provide a PR (and have no dev environment setup, atm). Would you mind having a look at this issue? I will provide additional logs and do testing as required, of course.

closed 2023-09-22 by BlackZork

BlackZork, 2023-09-22:
> Write holding registers to both components for starting/stopping filtering, setting pump motor speed, etc. I have the special need to write a single register value using Function 16 (0x10, [...]

That is interesting. Unfortunately mqmgateway uses write_multiple_holding_register only when number of register to write is > 1. Do do this you have to specify `count: (2 or more)` in config for command topic.

I am not sure what you mean at the end, you have workaround for this problem? Only one I can think of is to read at least two adjacent registers and write `["0","(last value]"`.

dw-1983, 2023-09-22:
Thanks for the fast fix! I see there is a docker pipeline set up, so probably the docker repo will auto-update at some time later?

Regarding my custom write_multiple case:
- Yes, my workaround currently is to write the target and the following register.
- I have seen implementations where you have a flag "write_type" (or register_type, in your implementation), which could be set to holdings (plural) and force a write_multiple_holding_register despite having a register count of 1.

Currently that workaround is working fine, because the followup register holds a binary value (that states "having heating installed" or not), which is always static (0) for my setup.

BlackZork, 2023-09-22:
> Thanks for the fast fix! I see there is a docker pipeline set up, so probably the docker repo will auto-update at some time later?

Unfortunately I do not have docker image build setup yet. There is a pull request #28, but it is not merged yet.

dw-1983, 2023-09-28:
@BlackZork just fyi, I finally managed to set up a linux targeting c++ dev environment and able to build the binaries and a docker container. After starting up the service, everything worked out fine right away (mqtt connection, modbus register read/write). Especially the automatic register grouping is a huge boost of performance and minimizes the configuration effort on my integration path into Home Assistant. Incomparable to the native modbus implementation in HA (pymodbus-based).
Thanks again for your work!

BlackZork, 2023-09-28:
Good to know that it works for you, thanks :-)

git-developer, 2024-01-13:
FYI: [v1.2.0](https://github.com/BlackZork/mqmgateway/releases/tag/v1.2.0) is available as Docker image under `ghcr.io/git-developer/mqmgateway:1.2.0`

commit 606b38eb (2023-09-22): missing auth data in Mosquitto::connect, closes #30