BlackZork/mqmgateway #32: Values are sended over MQTT only when changed

MaJEk-0, 2024-01-05:
Values are sent over MQTT only when changed. Would it be possible to send data that has not changed?

BlackZork, 2024-01-05:
You mean to publish MQTT topic value after every modbus poll?

MaJEk-0, 2024-01-05:
yes, value are publish over MQTT only if are changed.

BlackZork, 2024-01-05:
This is by design. Why would you like to publish the same value over and over again?

MaJEk-0, 2024-01-05:
it is not necessary to send every sample, but for example every 10th or 100th. In my project, I measure values that change every pool, but some values change once a day or a week. Here it would be useful for me to send the current status periodically.

pki791, 2024-01-05:
Possibly You should use retain on Your MQTT server, this will allow to see the stored last value. Or switch to modbus4mqtt :)

BlackZork, 2024-01-05:
> it is not necessary to send every sample, but for example every 10th or 100th. In my project, I measure values that change every pool, but some values change once a day or a week. Here it would be useful for me to send the current status periodically.

Values from modbus devices published by mqmgateway have retain flag set. After your client subscribe to MQTT topic, it should always get current value no matter when it was published.

MaJEk-0, 2024-01-05:
Thanks for the information, it was more about improving MQTT for data with a small period of change..

closed 2024-01-05 by BlackZork

BlackZork, 2024-08-26:
This feature was requested from other users too and I've added it in current master. See #56 for details.