BlackZork/mqmgateway #48: Error processing network traffic: Invalid function arguments provided

mczerski, 2024-05-17:
I get this error when trying to use modmqttd and have no clue what is wrong.
I'm using mosquitto as broker started like that:
```
docker run -p 1883:1883 -d --rm eclipse-mosquitto
```
modmqttd config:
```
modmqttd:
  converter_search_path:
    - /usr/lib/modmqttd
  converter_plugins:
    - stdconv.so
modbus:
  networks:
    - name: rtutest
      device: /dev/ttyUSB0
      baud: 115200
      parity: N
      data_bit: 8
      stop_bit: 1
mqtt:
  client_id: modbus
  #refresh: 1ms
  broker:
    host: localhost
  objects:
    - topic: test
      state:
        name: led
        register: rtutest.1.1
        register_type: coil
```
stdout of modmqttd:
```
2024-May-17 08:13:40.140977: [INFO ] modmqttd is starting
2024-May-17 08:13:40.141765: [INFO ] Added converter plugin std
2024-May-17 08:13:40.143092: [INFO ] Connecting to localhost:1883
2024-May-17 08:13:40.143256: [INFO ] Creating RTU context: /dev/ttyUSB0, 115200-8N1
2024-May-17 08:13:40.143313: [INFO ] Response timeout set to 500ms
2024-May-17 08:13:40.143517: [INFO ] modbus: connecting
2024-May-17 08:13:40.143675: [INFO ] Disconnected from mqtt broker, code:The connection was lost.
2024-May-17 08:13:40.143731: [INFO ] reconnecting to mqtt broker
2024-May-17 08:13:40.211149: [INFO ] modbus: connected
2024-May-17 08:13:40.211288: [INFO ] Waiting for mqtt network to become online
2024-May-17 08:13:40.211309: [INFO ] Connecting to localhost:1883
2024-May-17 08:13:40.211351: [INFO ] Waiting for mqtt network to become online
2024-May-17 08:13:40.211602: [ERROR] Error processing network traffic: Invalid function arguments provided.
2024-May-17 08:13:40.211924: [INFO ] Disconnected from mqtt broker, code:The connection was lost.
2024-May-17 08:13:40.212069: [INFO ] reconnecting to mqtt broker
2024-May-17 08:13:40.212121: [INFO ] Waiting for mqtt network to become online
```

BlackZork, 2024-05-17:
Started mosquitto from docker as in your example and got the same connection problem. I think you need to configure mosquitto to allow connections from remote network. docker logs shows:

```
[zork@archdevel ~]$ docker logs 9f5c771b5956
1715930388: mosquitto version 2.0.18 starting
1715930388: Config loaded from /mosquitto/config/mosquitto.conf.
1715930388: Starting in local only mode. Connections will only be possible from clients running on this machine.
1715930388: Create a configuration file which defines a listener to allow remote access.
1715930388: For more details see https://mosquitto.org/documentation/authentication-methods/
```

mczerski, 2024-05-17:
yeah, thats it. got it working with network=host.
I must say that I'm very impressed how fast modmqttd talks to my usb-uart adapter. I am able to reach around 500 request/response cycles per second for reading single coil.

closed 2024-05-17 by mczerski

BlackZork, 2024-05-17:
I am glad that it worked :-) 

Yes, speed is a priority here. I am using this gateway for controlling light switches by polling registers that holds switch counters. A 0.3s or slower for full round-trip (modbus poll -> MQTT publish counter -> process -> publish light toggle -> modbus write) starts to feel clumsy :-).

For RTU with single slave it is very fast. Things starts to get complicated when there are multiple slaves from different vendors. You may need to add some delays in this case.

mczerski, 2024-05-17:
> Yes, speed is a priority here. I am using this gateway for controlling light switches by polling registers that holds switch counters.

This is exactly what I want to do :-) right now I just want to make a proof of concept but it seems like I found a proof already :-D glad to know I'm heading the right direction.

I will be using my own hardware for that so I'm not worried by clumsy firmware in modbus slaves.