BlackZork/mqmgateway #56: Feature Request: Add the ability to publish in either direction even when their is no change in register value

JayHerpin, 2024-06-25:
There are a few use cases when this is beneficial. 

Modbus -> MQTT:

It is sometimes useful to have the timestamp of values updating those values are not changing. Consider the simple task so plotting the value of some value vs time. In the current implementation, doing this would require obersving changes in the "availability" of a particular value and assuming that the value had not changed because it still says it is available and no update had been received. While doable, it needlessly complicates the issue.

MQTT -> Modbus:

Consider the case where a register may have multiple writers or can otherwise value outside the view of mqmgateway. Given as mqmgateway will presumably always be used within the scope of some larger application or system, if that application or system wants to forcefully set some particular register to some value, a common way to do this might be to write that periodically. However, given that mqmgateway will discard operations that appear to have no effect, subsequent writes of the same value become ineffective. If the register in question where to change without that change being passed through mqmgateway then the larger application would effectively be unable to enforce its desired state.

BlackZork, 2024-06-25:
As for MQTT -> Modbus write commands are always executed. If you see other behaviour, then it is a bug.

For Modbus -> MQTT I agree, it should be configurable on mqtt level and topic level. It may be also useful for debugging purposes.

JayHerpin, 2024-06-25:
I'll double check the mqtt->modbus direction and file a bug if its not writing.

BlackZork, 2024-08-23:
Added in 1ef0307 (upcoming 2.10.0)

closed 2024-08-23 by BlackZork