BlackZork/mqmgateway #66: configuraition issue

mczerski, 2024-09-15:
I/m using v2.8.1
This is my config:
```
modmqttd:
  converter_search_path:
    - /usr/lib/modmqttd
  converter_plugins:
    - stdconv.so
modbus:
  networks:
    - name: rtugpio
      response_timeout: 50ms
      response_data_timeout: 50ms
      device: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
      # serial port parameters
      baud: 9600
      parity: N
      data_bit: 8
      stop_bit: 1
      delay_before_command: 10ms
mqtt:
  client_id: modbus
  refresh: 100ms
  broker:
    host: localhost
  objects:
    - topic: modbus/gpio
      network: rtugpio
      slave: 1
      state:
        register: 1
        register_type: bit
        count: 16
    - topic: modbus/kitchen_motion
      network: rtugpio
      slave: 2
      state:
        register: 2
        register_type: holding
    - topic: modbus/livingroom_led_dimmer/level
      network: rtugpio
      slave: 3
      state:
        register: 1
        register_type: input
      commands:
        - name: set
          register: 1
          register_type: holding
    - topic: modbus/livingroom_scene/1
      network: rtugpio
      slave: 3
      state:
        register: 2
        register_type: input
    - topic: modbus/livingroom_scene/2
      network: rtugpio
      slave: 3
      state:
        register: 3
        register_type: input
```

The problem is that under topics modbus/livingroom_scene/1 and modbus/livingroom_scene/2 there is always value of the register 1 and not the configured register for each topic.
What I want to achive is to have values of the first 3 input register of the slave 3 on separate mqtt topics. Should I write my config in different way ?

BlackZork, 2024-09-15:
I would leave slave 3 topics only, set refresh to 1sec, run modmqttd with --loglevel=5 (debug) or --loglevel=6 (trace) 2 seconds and look into logs what is read from slaves and published to mqtt broker. 

If it does not help, then post logs here.

mczerski, 2024-09-15:
So I concluded that my config is ok and I did not even try to launch suggested debug commands. The problem was on the slave device side (it was not took register address into account just the register count).

closed 2024-09-15 by mczerski

BlackZork, 2024-09-16:
Ok, thanks. 

Consider adding a poll group for slave 3 (register 1, count 3). It will pull data once for all three topics then.