BlackZork/mqmgateway #67: Publish timestamp along with modbus data

dakhnod, 2024-09-25:
Is there a way of publishing a timestamp along with the modbus values via mqtt?
So that the MQTT payload looks something like
```json
{
  "irradiance":10,
  "temp_cell_leg_1":517,
  "temp_ext_leg_1":0,
  "wind_speed":0,
  "reserved":0,
  "temp_cell_leg_2":1267,
  "temp_ext_leg_2":0,
  "temp_cell":26.7, 
  "timestamp": 1727276050676
}
```

BlackZork, 2024-09-25:
Not yet. What kind of timestamp do you need? Seconds since epoch in local time zone? Do you need milliseconds/microseconds precision?

dakhnod, 2024-09-25:
Hmm...millis should suffice. How about something like a dummy register with `converter: std.epoch_millis()` or something?

Also, is it possible to provide a list of converters for stacking them?

BlackZork, 2024-09-25:
Yes, the simplest method would be to add std.timestamp(format="") with strftime() like formats extended with milli and microsecond modifiers. 

Converter needs list of register values as input and returns [MqttValue](https://github.com/BlackZork/mqmgateway/blob/04f618bd287ab2c6e3d1c41bd40a1b91ee84ccb4/libmodmqttconv/mqttvalue.hpp#L9) as output - this interface makes stacking impossible. If you need to perform multiple conversions or math it is better to use expr plugin.

I have not tested it, but it should be possible to add a timestamp converter to dict member like this:

```
state:
  registers:
    - name: my_value
      register: 1
      register_type: input
    - name: ts
      register: 1
      register_type: input
      converter: std.timestamp()
```

This should output ```{ "my_value": 12, "ts": 1242424244 }. Still a hack, because timestamp will be updated on every mqtt payload creation, not when the register 1 was last read as I would expect.

dakhnod, 2024-09-25:
Okay, thanks for the suggestion! It would indeed be better though if the timestamp was set when the group of registers was read...

BlackZork, 2024-09-25:
It will work in most cases, because mqtt payload creation is triggered when at least one related register value is changed (or every poll if you set `publish_mode` to `every_poll`).

It will not work if you specify different registers for data and timestamp and set different refresh times in modbus section for them like this:

```
modbus:
  networks:
    - name: tcptest
      address: localhost
      port: 501
      slaves:
        - address: 1
          poll_groups:
            - register: 1
              register_type: input
              refresh: 10ms
            - register: 2
              register_type: input
              refresh: 1s
```

and in state:

```state:
  registers:
    - name: my_value
      register: tcptest.1.1
      register_type: input
    - name: ts
      register: tcptest.1.2
      register_type: input
      converter: std.timestamp()
```

Mqtt payload will be generated every 10ms, so `ts` will be updated every 10ms.

No idea if this makes any sense in real life, but I think one would expect timestamp to be updated after register 2 was read (every 1s), not any register used to create mqtt payload.

dakhnod, 2024-09-26:
Is `std.timestamp()` implemented? Because it doesn't work just like thet: `Converter std.timestamp not found`

BlackZork, 2024-09-26:
> Is `std.timestamp()` implemented? Because it doesn't work just like that: `Converter std.timestamp not found`

No, I did not do any work on it. You can create PR for std plugin it or create you own plugin with a custom converter - example code is [here](https://github.com/BlackZork/mqmgateway#adding-custom-converters)

dakhnod, 2024-09-26:
Could you please quickly explain why there is a refresh period in the networks section, and another in the mqtt object section? What if they are different? What is I set neither?

dakhnod, 2024-09-26:
[done](https://github.com/BlackZork/mqmgateway/pull/69)

BlackZork, 2024-09-26:
You can set refresh on network, slave, topic and register level. Lower level setting will override higher level setting.

On network level default refresh is 5s if not defined.