BlackZork/mqmgateway #68: MQTT connection refused with NATS MQTT broker

dakhnod, 2024-09-26:
I am using a NATS instance as my MQTT broker, and mqmgateway is connecting to it, to then throw the error "connection refused".
Do you have any Idea of a workaround or fix for that?

```
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [DEBUG] Client modbus sending CONNECT
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [DEBUG] Client modbus received CONNACK (4)
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [INFO ] Connection established
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [INFO ] Mqtt connected, sending subscriptions…
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [INFO ] Mqtt ready to process messages
Sep 26 12:34:46 nixos-daniel modmqttd[107815]: [INFO ] Disconnected from mqtt broker, code:The connection was refused.
```

dakhnod, 2024-09-26:
Okay, thanks anyways!

dakhnod, 2024-09-26:
Oh, I know what it is, mqm sends a username and password although none are configured. Am I getting somethign wrong, or is there a way to disable that?

dakhnod, 2024-09-26:
Got it. 
Apparently, NATS does accept any username/password, excluding an empty username and password.
Mqm uses an empty username and password, which exactly triggers NATS to refuse the connection.

Setting a username and password in the config yaml allow mqm to connect to NATS via MQTT.

BlackZork, 2024-09-26:
Are you able to comment out mqtt user/pass [here](https://github.com/BlackZork/mqmgateway/blob/04f618bd287ab2c6e3d1c41bd40a1b91ee84ccb4/libmodmqttsrv/mosquitto.cpp#L103) ln 103 and 104, recompile and see if it helps?

closed 2024-09-26 by BlackZork

dakhnod, 2024-09-27:
Thank you! Should work, i'll get back to it