BlackZork/mqmgateway #78: device becomes unavailable with one unreadable register

dakhnod, 2025-02-07:
Again, huge thanks for your work! I really appreciate it!

If I understand right, if a single register from a devices state is unreadable, the whole device becomes unavailable and no register data is delivered in "every_poll" mode.

Is there a way to instead define an error value that will be delivered for an unredible register?

BlackZork, 2025-02-07:
> If I understand right, if a single register from a devices state is unreadable, the whole device becomes unavailable and no register data is delivered in "every_poll" mode.

This is correct per topic defined in configuration, not per device.

Topic state is treated as atomic value - you either have valid value or no value. If you need to monitor multiple properties just define multiple topics for each property. It is better to group registers in single topic if you know that you can read all of them (single device, single network).

"every_poll" mode is used when you want to monitor value and delivery using state topic only. If data feed stops, then there is something wrong. See [original feature request description](https://github.com/BlackZork/mqmgateway/issues/56).

closed 2025-02-11 by BlackZork

dakhnod, 2025-02-11:
Unfortunately, I am in a position where I need the all registers of interest in a single published object. What do you think of defining an error_value per register, which allows to publish said error_value when the register cannot be read, instead of discarding the whole object?

BlackZork, 2025-02-11:
This may be a good addition to mqmgateway in cases where this object is composed from data from multiple devices. Maybe I am wrong but for a single device in most cases you will always be able to read all registers or none of them. Much higher chance is that you will get bogus data from some of registers due to device malfunction.

Note that currenlty there is a hardcoded number of 3 tries before mqmgateway gives up and sends MsgRegisterReadError to MqttClient.