BlackZork/mqmgateway #83: Allow slave id 0

Trickfilm400, 2025-03-23:
Hi,

I have a device which only responds with a slave id of 0, but when trying to configure it, I get the following error: `[CRITICAL] config error(line 17): Invalid slave address range 0-0`.

Could you please allow using slave id 0? Could also be an extra flag, because its the broadcast slave id - otherwise this great project isn't usable for me

Thank you

BlackZork, 2025-03-23:
Fixed in enable_slave_0 branch. I do not have a setup to test it - are you able to compile and test mqmgateway from this branch?

Trickfilm400, 2025-03-24:
Hi,
thanks for the patch,
I've built the docker container for the branch and got the following console output:
```
2025-Mar-24 20:03:24.289672: [INFO ] modmqttd is starting
2025-Mar-24 20:03:24.292680: [INFO ] Added converter plugin std
2025-Mar-24 20:03:24.293066: [INFO ] Connecting to 192.168.XX.XX:502
2025-Mar-24 20:03:24.293181: [INFO ] Response timeout set to 500ms
2025-Mar-24 20:03:24.293394: [INFO ] modbus: connecting
2025-Mar-24 20:03:24.295496: [INFO ] modbus: connected
2025-Mar-24 20:03:24.295591: [INFO ] Waiting for mqtt network to become online
2025-Mar-24 20:03:24.295558: [INFO ] Connecting to <emqx>:1883
2025-Mar-24 20:03:24.295812: [INFO ] Waiting for mqtt network to become online
2025-Mar-24 20:03:24.309598: [INFO ] Connection established
2025-Mar-24 20:03:24.309728: [INFO ] Mqtt connected, sending subscriptions…
2025-Mar-24 20:03:24.309839: [INFO ] Mqtt ready to process messages
2025-Mar-24 20:03:24.325210: [ERROR] 1 error(s) when reading register 0.30368, last error: libmodbus: read fn 30368 failed with return code -1: Illegal data value
2025-Mar-24 20:03:24.330244: [INFO ] Initial poll done in 34ms
```

I'm using the following yaml:

```yaml
#<stuff before>
  objects:
    - topic: smarthome/test/a
      refresh: 5s
      # override default, publish after every modbus read call.
      publish_mode: every_poll
      state:
        name: a
        # also tested with decimal value
        register: tcptest.0.0x76A0
        register_type: holding
        count: 1
        #register_type: input
        refresh: 2s
```

Here is a screenshot of the working config with this desktop test client: (ignore the disconnected state)

[image]

Did I configure the yaml incorrect? I'm a bit confused

BlackZork, 2025-03-24:
I would suggest doing the following:

1. Make sure that only one master is quering your slave at once.
2. run mqmgateway with DEBUG or TRACE log level, set refresh to long value (10s or more) to capture only one read call.
3. try to use mbpoll to query your slave - it uses the same libmodbus library under the hood, if you are able to read your slave with mbpoll then mqmgateway should work or I surely will be able to fix it to make it work.

Also, post full config and a log from a single read call here.

Trickfilm400, 2025-03-24:
1. It should only be one connection to the modbus tcp _server_ (as writtenbelow, my other modbus gateway and the desktop client works and I stopped it for this test here)
2. with the trace loglevel:

```
2025-Mar-24 21:25:16.703281: [INFO ] modmqttd is starting
2025-Mar-24 21:25:16.705541: [DEBUG] Checking "stdconv.so"
2025-Mar-24 21:25:16.705659: [DEBUG] Checking "/usr/lib/modmqttd/stdconv.so"
2025-Mar-24 21:25:16.705747: [DEBUG] Trying to load converter plugin from /usr/lib/modmqttd/stdconv.so
2025-Mar-24 21:25:16.706099: [INFO ] Added converter plugin std
2025-Mar-24 21:25:16.706269: [DEBUG] Broker configuration initialized
2025-Mar-24 21:25:16.706510: [DEBUG] Modbus thread started
2025-Mar-24 21:25:16.706657: [TRACE] Waiting for messages
2025-Mar-24 21:25:16.706806: [INFO ] Connecting to 192.168.XX.XX:502
2025-Mar-24 21:25:16.706954: [INFO ] Response timeout set to 500ms
2025-Mar-24 21:25:16.707082: [DEBUG] Watchdog initialized. Watch period set to 10s
2025-Mar-24 21:25:16.707252: [INFO ] modbus: connecting
2025-Mar-24 21:25:16.708167: [DEBUG] 1 modbus client(s) initialized
2025-Mar-24 21:25:16.708509: [DEBUG] processing object smarthome/test/a
2025-Mar-24 21:25:16.709623: [INFO ] modbus: connected
2025-Mar-24 21:25:16.709697: [DEBUG] Adding new register 0.30368 (1) type=3 refresh=10000 on network tcptest
2025-Mar-24 21:25:16.709750: [DEBUG] starting initial poll
2025-Mar-24 21:25:16.709823: [INFO ] Waiting for mqtt network to become online
2025-Mar-24 21:25:16.709798: [DEBUG] Min publish rate for smarthome/test/a/state set to 10000ms
2025-Mar-24 21:25:16.709949: [TRACE] Watchdog: current error period is 0ms
2025-Mar-24 21:25:16.710059: [TRACE] Waiting for messages
2025-Mar-24 21:25:16.710028: [DEBUG] object for topic smarthome/test/a created
2025-Mar-24 21:25:16.710220: [DEBUG] Finished reading mqtt object declarations
2025-Mar-24 21:25:16.710182: [INFO ] Waiting for mqtt network to become online
2025-Mar-24 21:25:16.710355: [TRACE] Watchdog: current error period is 0ms
2025-Mar-24 21:25:16.710324: [DEBUG] Adding new register 0.30368 (1) type=3 refresh=10000 on network tcptest
2025-Mar-24 21:25:16.710495: [TRACE] Waiting for messages
2025-Mar-24 21:25:16.710534: [DEBUG] Sending register specification to modbus thread for network tcptest
2025-Mar-24 21:25:16.710693: [DEBUG] Poll specification set, got 1 slaves,1 registers to poll:
2025-Mar-24 21:25:16.710787: [DEBUG] tcptest, slave 0, register 30368:3, count=1, poll every 10000ms, queue always, min f_delay 1500ms, min delay 0ms
2025-Mar-24 21:25:16.710750: [DEBUG] Performing initial connection to mqtt broker
2025-Mar-24 21:25:16.710923: [INFO ] Connecting to <emqx>:1883
2025-Mar-24 21:25:16.710905: [TRACE] Starting election for silence period 360000000004ms
2025-Mar-24 21:25:16.711141: [TRACE] Next register to poll set to 0.30368, commands_left=2
2025-Mar-24 21:25:16.711225: [DEBUG] starting initial poll
2025-Mar-24 21:25:26.784884: [TRACE] Register 0.30368 (0x0.0x76a0) added, last read 10000ms ago
2025-Mar-24 21:25:26.785032: [TRACE] Wait duration set to 10000ms as next poll for register 0.30368 (0x0.0x76a0)
2025-Mar-24 21:25:26.785116: [TRACE] Starting election for silence period 10001ms
2025-Mar-24 21:25:26.785199: [TRACE] Next register to poll set to 0.30368, commands_left=2
2025-Mar-24 21:25:26.785269: [TRACE] Scheduling 1 registers to execute, next schedule in 10000ms
2025-Mar-24 21:25:26.788712: [TRACE] Watchdog: current error period is 10078ms
2025-Mar-24 21:25:26.788794: [ERROR] Cannot execute any command in last 10s, reconnecting
2025-Mar-24 21:25:26.788912: [INFO ] modbus: connecting
2025-Mar-24 21:25:26.788993: [DEBUG] Client modbus sending PUBLISH (d0, q0, r1, m1, 'smarthome/test/a/availability', ... (1 bytes))
2025-Mar-24 21:25:26.790998: [INFO ] modbus: connected
2025-Mar-24 21:25:26.791107: [TRACE] Starting election for silence period 2ms
2025-Mar-24 21:25:26.791185: [TRACE] Next register to poll set to 0.30368, commands_left=2
2025-Mar-24 21:25:26.791267: [DEBUG] starting initial poll
2025-Mar-24 21:25:26.806213: [INFO ] Initial poll done in 15ms
2025-Mar-24 21:25:26.806246: [DEBUG] Client modbus sending PUBLISH (d0, q0, r1, m2, 'smarthome/test/a/availability', ... (1 bytes))
2025-Mar-24 21:25:26.806357: [TRACE] Watchdog: current error period is 15ms
2025-Mar-24 21:25:26.806429: [TRACE] Checking for incoming message
2025-Mar-24 21:25:26.806705: [TRACE] Watchdog: current error period is 15ms
2025-Mar-24 21:25:26.806781: [TRACE] Waiting for messages for  9978ms
2025-Mar-24 21:25:36.784434: [TRACE] Wait duration set to 20ms as next poll for register 0.30368 (0x0.0x76a0)
2025-Mar-24 21:25:36.784573: [TRACE] Scheduling 0 registers to execute, next schedule in 20ms
2025-Mar-24 21:25:36.784669: [TRACE] Watchdog: current error period is 9994ms
2025-Mar-24 21:25:36.784742: [TRACE] Waiting for messages for  20ms
2025-Mar-24 21:25:36.805884: [TRACE] Register 0.30368 (0x0.0x76a0) added, last read 10000ms ago
2025-Mar-24 21:25:36.806096: [TRACE] Wait duration set to 10000ms as next poll for register 0.30368 (0x0.0x76a0)
2025-Mar-24 21:25:36.806251: [TRACE] Starting election for silence period 10001ms
2025-Mar-24 21:25:36.806363: [TRACE] Next register to poll set to 0.30368, commands_left=2
2025-Mar-24 21:25:36.806438: [TRACE] Scheduling 1 registers to execute, next schedule in 10000ms
2025-Mar-24 21:25:36.809356: [TRACE] Watchdog: current error period is 10019ms
```

[... 83 lines cut ...]
-- Polling slave 0... Ctrl-C to stop)
[... cut ...]
Read output (holding) register failed: Illegal data value
--- 192.168.XX.XX poll statistics ---
4 frames transmitted, 0 received, 4 errors, 100.0% frame loss

everything was closed.
Have a nice day !

C:\Program Files (x86)\MBPoll>
```

so, the `mbpoll` doesn't seem to work either, if I used it correctly.

----

In the meantime I used another modbus/mqtt gateway: https://github.com/mazocode/modbus2mqtt
This works also without any issues

Does this information help you maybe? Or do you think it's an underlying issue with the `libmodbus` library?

BlackZork, 2025-03-24:
Not sure. Error code is strange - it is used for writing, not reading. Next option is to use packet sniffer and see what is the difference between packets sent by modbus2mqtt or other working tool and mbpoll.

BlackZork, 2025-03-24:
Looks  like this is it. MODBUS_TCP_SLAVE is defined as 0xff here:

https://github.com/stephane/libmodbus/blob/bb1344325025e59bb33144394bdacd8298283834/src/modbus-tcp.h#L35

Trickfilm400, 2025-03-25:
Yes, this resolved the issue:

```diff
- if (slaveId != 0)
-     modbus_set_slave(mCtx, slaveId);
- else
-     modbus_set_slave(mCtx, MODBUS_TCP_SLAVE);
+ modbus_set_slave(mCtx, 0);
```
The correct value is published: 

[image]

BlackZork, 2025-03-25:
Good :-)
This obviously will broke well-behaving slaves (0xff is a default slave id for TCP), so proper fix is needed for both cases.

BlackZork, 2025-03-26:
Apparently this MODBUS_TCP_SLAVE constant was impossible to use - there is no way to define topic without slave id.
I fixed this override for slave id=0 leaving this default for future changes (#84).

Please let me know if it works for you. Thanks!

Trickfilm400, 2025-03-26:
I applied the [c00445a](https://github.com/BlackZork/mqmgateway/commit/c00445aea5268cfb76b33378eee441774e5fb1d0) commit and it works correctly. Thanks

BlackZork, 2025-03-26:
Thanks, merged to master, official images for master branch should be rebuild shortly.

commit c00445ae (2025-03-26): Do not override slave with id=0 to 0xff. Fixes #83
closed 2025-03-26 by BlackZork