BlackZork/mqmgateway #86: High CPU load

nick-ua, 2025-04-06:
Hello. I am using docker image on orange pi zero (ARMv7). 

When I poll 1 coil with refresh: 10ms:
```
%CPU CMD
30.0   /usr/bin/modmqttd --config /etc/modmqttd/config.yaml
```

refresh: 100ms
```
%CPU CMD
 9.1 /usr/bin/modmqttd --config /etc/modmqttd/config.yaml
```

refresh: 1s
```
%CPU CMD
 1.6 /usr/bin/modmqttd --config /etc/modmqttd/config.yaml
```

I wrote simple test program that polls coils

refresh 10ms:
```
%CPU CMD
 1.0 ./test.bin
```

refresh 100ms:
```
%CPU CMD
 0.2 ./test.bin
```

refresh 1s:
```
%CPU CMD
 0.0 ./test.bin
```
Any chance to improve this? How I can help you?

[config.txt](https://github.com/user-attachments/files/19622222/config.txt)

[main_c.txt](https://github.com/user-attachments/files/19622283/main_c.txt)

BlackZork, 2025-04-06:
Hard to say without knowing exact cause. 

Both programs are run inside docker container? The first thing I would do is to remove docker overhead - run and compile mqmgateway directly.

You may try to decrease log level or even [turn off boost logging in source code](https://www.boost.org/doc/libs/1_63_0/libs/log/doc/html/boost/log/core.html#idp41743280-bb) and see if it helps.

nick-ua, 2025-04-07:
Only modmqttd was running inside container. Now I compiled modmqttd from sources and run without docker. Results are same. I added after  BOOST_LOG_SEV(*log, modmqttd::Log::info) << "modmqttd is starting"; in modmqttd/main.cpp :

```
        boost::log::core::get()->remove_all_sinks();
        boost::log::core::get()->set_logging_enabled(false);
```
Nothing changed drastically.

I will try to  use profiler.

BlackZork, 2025-04-07:
modqmttd will create 3 threads for your config. I assume that modbus thread is causing this CPU load. 
The code that selects registers that needs to be read ([ModbusScheduler](https://github.com/BlackZork/mqmgateway/blob/master/libmodmqttsrv/modbus_scheduler.hpp)) and do actual polling and command execution ([ModbusExecutor](https://github.com/BlackZork/mqmgateway/blob/master/libmodmqttsrv/modbus_executor.hpp)) is much more complicated than your example program, but idea is similar: ModbusExecutor issues all needed libmodbus I/O and then returns a time period how long thread should sleep.

Thread sleeps on incoming messages queue [here](https://github.com/BlackZork/mqmgateway/blob/master/libmodmqttsrv/modbus_thread.cpp#L274).

Either some idleWaitDuration calcuations are wrong ([logged here](https://github.com/BlackZork/mqmgateway/blob/master/libmodmqttsrv/modbus_thread.cpp#L273), you should see them when running with TRACE log level), or there is some problem with wait_dequeue_timed. In the second case you may try to update [readerwriterqueue](https://github.com/BlackZork/mqmgateway/tree/master/readerwriterqueue) from https://github.com/cameron314/readerwriterqueue and see if this helps.

BlackZork, 2025-04-07:
You can also attach a log with TRACE log level (--loglevel=6) here from a 1-3 sec session with 100ms poll time period. Maybe there is something suspicious in the log.

BlackZork, 2025-04-07:
I do not see any performance problems. Thread is sleeping 100ms in two parts between modbus calls. Not ideal, but should not cause high CPU usage. Maybe wait_dequeue_timed() is CPU intensive? You may replace it with thread::sleep() and see if it helps.

BlackZork, 2025-04-07:
> You may add thread::sleep() before wait_dequeue_timed and see if it helps.

I mean replace ```wait_dequeue_timed(duration)``` with ```thread::sleep(duration)``` and ```try_deque()```.

nick-ua, 2025-04-07:
libmodmqttsrv/modbus_thread.cpp :
```
BOOST_LOG_SEV(log, Log::trace) << constructIdleWaitMessage(idleWaitDuration);
int64_t duration_ms = std::chrono::duration_cast < std::chrono::milliseconds > (idleWaitDuration).count();
if (idleWaitDuration != std::chrono::steady_clock::duration::max()) {
  std::this_thread::sleep_for(std::chrono::milliseconds(duration_ms));
}
if (!mToModbusQueue.try_dequeue(item)) {
  continue;
}
```
before:

[image]

after I changed to  std::this_thread::sleep_for :

[image]

and disabled logs :

[image]

I use `perf record -g ./modmqttd --config ~/config.yaml` and `perf report`

Now with `publish_mode: every_poll` and `refresh: 10ms ` :
```
%CPU CMD
13.3 ./modmqttd/modmqttd --config ~/config.yaml

```

`publish_mode: on_change` and `refresh: 10ms ` :
```
%CPU CMD
 9.3 ./modmqttd/modmqttd --config ~/config.yaml
```

`publish_mode: on_change` and `refresh: 100ms ` :
```
%CPU CMD
 2.3 ./modmqttd/modmqttd --config ~/config.yaml
```

nick-ua, 2025-04-07:
🤔 libmodmqttsrv/modbus_thread.cpp / ModbusThread::run() updated version:
```
BOOST_LOG_SEV(log, Log::trace) << constructIdleWaitMessage(idleWaitDuration);

if (idleWaitDuration != std::chrono::steady_clock::duration::max()) {
  int64_t duration_ms = std::chrono::duration_cast < std::chrono::milliseconds > (idleWaitDuration).count();
  std::this_thread::sleep_for(std::chrono::milliseconds(duration_ms));
  if (!mToModbusQueue.try_dequeue(item)) {
    continue;
  }
} else {
  if (!mToModbusQueue.wait_dequeue_timed(item, idleWaitDuration))
    continue;
}
```

BlackZork, 2025-04-07:
Many thanks for your work!

I think it would be good to add --log-level=0 that will turn off all logging. That is the easy part.

```std::this_thread::sleep_for``` is not a good solution - it will block all inbound communication when waiting for next poll. This will delay all commands sent from mqtt (modbus write requests), mqtt disconnections and other events.

I may try to use std::binary_semaphore for inter-thread communication, it may or may not be faster than this lightweight semaphore implmentations from readerwriterqueue.

BlackZork, 2025-04-07:
Looking at the https://github.com/BlackZork/mqmgateway/blob/master/readerwriterqueue/atomicops.h it seems that on ARM_7 __unix__ is not defined so it uses class LightweightSemaphore implementation instead of Semaphore wrapper.

You may try to force #define __unix__ in https://github.com/BlackZork/mqmgateway/blob/master/readerwriterqueue/atomicops.h, recompile and check if results are the same.

This looks like a bug in atomicops.h, it should pick Linux/posix wrapper instead of soft emulated version.

BlackZork, 2025-04-07:
> 
> This looks like a bug in atomicops.h, it should pick Linux/posix wrapper instead of soft emulated version.

Sorry, I misinterpreted atomicops.h, it uses POSIX semaphore implementation in LightweightSemaphore class, so it already uses semaphores to sync queue enque/deque operations. I am currently out of ideas how to spedd it up without slowing down input message processing.

BlackZork, 2025-04-08:
Looking at LightweightSemaphore a little bit futher I think that we can save some CPU cycles if we add external semaphore that will not have this "[spinning feature](https://github.com/BlackZork/mqmgateway/blob/c00445aea5268cfb76b33378eee441774e5fb1d0/readerwriterqueue/atomicops.h#L577)" - a 10000 iteration loop that check atomic value for changes.

I think that you can test it by setting this spinning counter to 0 [in this line ](https://github.com/BlackZork/mqmgateway/blob/master/readerwriterqueue/atomicops.h#L576)

BlackZork, 2025-04-08:
I do not think that nanomsg or other high level queue systems will be faster than ReaderWriterQueue. Event driven approach means to drop all threads and use non-blocking I/O everywhere. Supported by mosquitto, but not possible with libmodbus. Even if there is any async C modbus libary I do not even want to think how much work would be to convert all the code from threaded to async :-)

As you can see in your first screen 10000 spinning loop is causing most of the load (```weak_atomic>int>::load```). If you set it to 0 you should get a similar performance as with thread::sleep_for() approach. It would be great if you can confirm it on your hardware.

nick-ua, 2025-04-08:
Now with `refresh: 100ms` and `publish_mode: on_change`  CPU load ~ 1.2 - 1.5 % . Thank you!!!

closed 2025-04-08 by BlackZork

BlackZork, 2025-04-08:
I updated readerwriterqueue from current master ([16b48ae](https://github.com/cameron314/readerwriterqueue/tree/16b48ae)) and commented out spinning code. New docker images are already available. 

Many thanks for your time for testing, profiling.... and a caffee of course! 😃

commit 95a202ea (2025-04-08): Disable initial spinning in LightweightSemaphore. Closes #86