BlackZork/mqmgateway #87: Issues using std.multiply converter

dakhnod, 2025-04-11:
When running this minimal config
```
modmqttd:
  converter_search_path:
    - build/stdconv
    - build/exprconv
  converter_plugins:
    - stdconv.so
    - exprconv.so
modbus:
  networks:
    - name: test
      address: 127.0.0.1
      port: 5020
mqtt:
  client_id: modbus
  refresh: 3s
  publish_mode: every_poll
  broker:
    host: 127.0.0.1
    port: 1883
  objects:
    - topic: v1/test
      network: test
      slave: 1
      state:
        - register: "0x8724"
          register_type: holding
          name: value
          converter: std.multiply(10)
```

I am getting this error:
```
modmqttd: /usr/include/rapidjson/internal/dtoa.h:217: char* rapidjson::internal::dtoa(double, char*, int): Assertion `maxDecimalPlaces >= 1' failed.
```

Using `expr.evaluate(R0 * 10)` instead works, weird...

In any case, thanks for your work!

dakhnod, 2025-04-11:
This seems to be caused by `SingleArgMathConverter` specifiying a precision of 0, which the assert catches...

dakhnod, 2025-04-11:
I figured out why this never caused me issues so far;
I have only been using `std.multiply` on nixOS, where RapidJSON wasn't present, and hence no assert fired.

On ubuntu with RapidJSON this fails...

BlackZork, 2025-04-11:
Confirmed in [39bcc16](https://github.com/BlackZork/mqmgateway/commit/39bcc16f2b59be6e11c95ab0953edd0341389f82), not sure what you mean about RapidJSON - it is required.

Your config should output a single element json array.

dakhnod, 2025-04-11:
Okay, the config I posted works with a different converter. Not sure I understand you fully

BlackZork, 2025-04-11:
Fixed in [8c58197](https://github.com/BlackZork/mqmgateway/commit/8c5819726fe6d8fe4c73e0bf6094e3b33da8ab2b).

```std.multiply``` outputs MqttValue with mPrecision forced to 0
RapidJSON needs decimal point precision >= 1 when formatting double value. 

I fixed this exact case, but more tests are needed before merge to master.

dakhnod, 2025-04-11:
Ah, sorry, understood. It is unmerged, yet.

commit 8c581972 (2025-04-11): convert to int if SingleArgMathConverter declares precision=0, fixes #87. Fallback to 6 decimal places if precision is not set.
closed 2025-04-11 by BlackZork